at-wat/mcl_3dl #9: Error occurred when I do: rosrun mcl_3dl mcl_3dl

oya123, 2016-12-26:
After I "rosrun mcl_3dl mcl_3dl", there was occurred segment fault.
And I set breakpoint to find the position where cause this error. But I found that the error occurred before it going into main().
I don't know why it happened and hope you can help me.
Thanks for all.
my system is: Ubuntu 14.04, ROS indigo

at-wat, 2016-12-26:
Have you passed -g option to gcc?
Adding following code to CMakeLists.txt might help you to debug.
```
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -g -O0")
```

oya123, 2016-12-27:
I have added the code to my CMakeLists.txt, but the error has also occured.
My CMakeLists.txt 
`cmake_minimum_required(VERSION 2.8.3)
project(mcl_3dl)

find_package(catkin REQUIRED COMPONENTS
	roscpp std_msgs message_generation sensor_msgs pcl_ros
)

find_package(PCL 1.7 REQUIRED)

catkin_package(
	CATKIN_DEPENDS roscpp std_msgs sensor_msgs pcl_ros
	DEPENDS system_lib
)

include(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
if(COMPILER_SUPPORTS_CXX11)
	set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -g -O0")
elseif(COMPILER_SUPPORTS_CXX0X)
	set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
else()
	message(STATUS "The compiler ${CMAKE_CXX_COMPILER} has no C++11 support. Please use a different C++ compiler.")
endif()

include_directories(${catkin_INCLUDE_DIRS} ${PCL_INCLUDE_DIR} src)
link_directories(${PCL_LIBRARY_DIR})
add_definitions(${PCL_DEFINITIONS})

add_executable(mcl_3dl src/mcl_3dl.cpp)
target_link_libraries(mcl_3dl ${catkin_LIBRARIES} ${PCL_LIBRARIES})`

at-wat, 2016-12-27:
The code I suggested embeds debug information; It doesn't fix error.
Please run the node compiled with '-g' option under the debugger to find position of the error.

oya123, 2016-12-28:
Thanks for your help, I'll try it again.

at-wat, 2016-12-28:
Usually, I use fedora. 
I just tried this on Ubuntu 14.04 with ROS Indigo on my Intel NUC (Core i7 model), and it worked.

It is appreciated if you could give me some debugging results.

oya123, 2017-01-03:
This is my compilation information, but I didn't find any position could lead to error.
`Base path: /home/ouyangqiuping/mcl_3d
Source space: /home/ouyangqiuping/mcl_3d/src
Build space: /home/ouyangqiuping/mcl_3d/build
Devel space: /home/ouyangqiuping/mcl_3d/devel
Install space: /home/ouyangqiuping/mcl_3d/install
####
#### Running command: "cmake /home/ouyangqiuping/mcl_3d/src -DCATKIN_DEVEL_PREFIX=/home/ouyangqiuping/mcl_3d/devel -DCMAKE_INSTALL_PREFIX=/home/ouyangqiuping/mcl_3d/install -G Unix Makefiles" in "/home/ouyangqiuping/mcl_3d/build"
####
-- The C compiler identification is GNU 4.9.4
-- The CXX compiler identification is GNU 4.9.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /home/ouyangqiuping/mcl_3d/devel
-- Using CMAKE_PREFIX_PATH: /home/ouyangqiuping/mcl_3d/devel;/home/ouyangqiuping/lsd_slam_catkin/devel;/opt/ros/indigo
-- This workspace overlays: /home/ouyangqiuping/mcl_3d/devel;/home/ouyangqiuping/lsd_slam_catkin/devel;/opt/ros/indigo
-- Found PythonInterp: /home/ouyangqiuping/.venv/python2.7/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /home/ouyangqiuping/.venv/python2.7/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ouyangqiuping/mcl_3d/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - mcl_3dl
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'mcl_3dl'
-- ==> add_subdirectory(mcl_3dl-master)
-- Using these message generators: gencpp;geneus;genlisp;genpy
-- checking for module 'eigen3'
--   found eigen3, version 3.2.8
-- Found eigen: /usr/local/include/eigen3  
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found openni: /usr/lib/libOpenNI.so  
-- Found libusb-1.0: /usr/include  
-- checking for module 'flann'
--   found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- looking for PCL_COMMON
-- Found PCL_COMMON: /usr/lib/libpcl_common.so  
-- looking for PCL_OCTREE
-- Found PCL_OCTREE: /usr/lib/libpcl_octree.so  
-- looking for PCL_IO
-- Found PCL_IO: /usr/lib/libpcl_io.so  
-- looking for PCL_KDTREE
[... 20 lines cut ...]
-- Found PCL_REGISTRATION: /usr/lib/libpcl_registration.so  
-- looking for PCL_RECOGNITION
-- Found PCL_RECOGNITION: /usr/lib/libpcl_recognition.so  
-- looking for PCL_SURFACE
-- Found PCL_SURFACE: /usr/lib/libpcl_surface.so  
-- looking for PCL_PEOPLE
-- Found PCL_PEOPLE: /usr/lib/libpcl_people.so  
-- looking for PCL_TRACKING
-- Found PCL_TRACKING: /usr/lib/libpcl_tracking.so  
-- looking for PCL_APPS
-- Found PCL_APPS: /usr/lib/libpcl_apps.so  
-- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/libpcl_common.so;debug;/usr/lib/libpcl_common.so;optimized;/usr/lib/libpcl_octree.so;debug;/usr/lib/libpcl_octree.so;/usr/lib/libOpenNI.so;vtkCommon;vtkRendering;vtkHybrid;vtkCharts;optimized;/usr/lib/libpcl_io.so;debug;/usr/lib/libpcl_io.so;optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;optimized;/usr/lib/libpcl_kdtree.so;debug;/usr/lib/libpcl_kdtree.so;optimized;/usr/lib/libpcl_search.so;debug;/usr/lib/libpcl_search.so;optimized;/usr/lib/libpcl_sample_consensus.so;debug;/usr/lib/libpcl_sample_consensus.so;optimized;/usr/lib/libpcl_filters.so;debug;/usr/lib/libpcl_filters.so;optimized;/usr/lib/libpcl_features.so;debug;/usr/lib/libpcl_features.so;optimized;/usr/lib/libpcl_keypoints.so;debug;/usr/lib/libpcl_keypoints.so;optimized;/usr/lib/libpcl_segmentation.so;debug;/usr/lib/libpcl_segmentation.so;optimized;/usr/lib/libpcl_visualization.so;debug;/usr/lib/libpcl_visualization.so;optimized;/usr/lib/libpcl_outofcore.so;debug;/usr/lib/libpcl_outofcore.so;optimized;/usr/lib/libpcl_registration.so;debug;/usr/lib/libpcl_registration.so;optimized;/usr/lib/libpcl_recognition.so;debug;/usr/lib/libpcl_recognition.so;optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so;optimized;/usr/lib/libpcl_surface.so;debug;/usr/lib/libpcl_surface.so;optimized;/usr/lib/libpcl_people.so;debug;/usr/lib/libpcl_people.so;optimized;/usr/lib/libpcl_tracking.so;debug;/usr/lib/libpcl_tracking.so;optimized;/usr/lib/libpcl_apps.so;debug;/usr/lib/libpcl_apps.so;/usr/lib/x86 [... cut ...]
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ouyangqiuping/mcl_3d/build
####
#### Running command: "make -j4 -l4" in "/home/ouyangqiuping/mcl_3d/build"
####
Scanning dependencies of target mcl_3dl
[100%] Building CXX object mcl_3dl-master/CMakeFiles/mcl_3dl.dir/src/mcl_3dl.cpp.o
Linking CXX executable /home/ouyangqiuping/mcl_3d/devel/lib/mcl_3dl/mcl_3dl`

at-wat, 2017-01-03:
Build seems OK.
Have you tried debugger?

Please post full gdb result of following commands.
```
$ gdb ~/mcl_3d/devel/lib/mcl_3dl/mcl_3dl
(gdb) run

* Some error (segmentation fault?) occurs *

(gdb) info stack
```

oya123, 2017-01-03:
The result is followed:

`GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from mcl_3dl...done.
(gdb) run
Starting program: /home/ouyangqiuping/mcl_3d/devel/lib/mcl_3dl/mcl_3dl 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff4478ae0 in boost::math::lanczos::lanczos_initializer<boost::math::lanczos::lanczos17m64, long double>::init::init() ()
   from /usr/lib/libpcl_sample_consensus.so.1.7
(gdb) `

at-wat, 2017-01-03:
Please post `info stack` result.

oya123, 2017-01-03:
(gdb) info stack
#0  0x00007ffff4478ae0 in boost::math::lanczos::lanczos_initializer<boost::math::lanczos::lanczos17m64, long double>::init::init() ()
   from /usr/lib/libpcl_sample_consensus.so.1.7
#1  0x00007ffff44505ce in ?? () from /usr/lib/libpcl_sample_consensus.so.1.7
#2  0x00007ffff7dea10a in call_init (l=<optimized out>, argc=argc@entry=1, 
    argv=argv@entry=0x7fffffffcdf8, env=env@entry=0x7fffffffce08)
    at dl-init.c:78
#3  0x00007ffff7dea1f3 in call_init (env=<optimized out>, 
    argv=<optimized out>, argc=<optimized out>, l=<optimized out>)
    at dl-init.c:36
#4  _dl_init (main_map=0x7ffff7ffe1c8, argc=1, argv=0x7fffffffcdf8, 
    env=0x7fffffffce08) at dl-init.c:126
#5  0x00007ffff7ddb30a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#6  0x0000000000000001 in ?? ()
#7  0x00007fffffffd2a2 in ?? ()
#8  0x0000000000000000 in ?? ()

at-wat, 2017-01-03:
This seems a bug in PCL 1.7.
* http://answers.ros.org/question/194699/segmentation-fault-when-using-correspondencerejectorsampleconsensus/
* https://github.com/PointCloudLibrary/pcl/issues/619

Answer in the following thread might fix this on PCL 1.7.
* http://stackoverflow.com/questions/26346690/debug-seg-fault-in-boostmathlanczos-with-libpcl-surface

oya123, 2017-01-03:
Thanks a lot! I solve this issue as this answer said:

oya123, 2017-01-03:
Thanks a lot! I solved this issue as this answer said:
http://stackoverflow.com/questions/26346690/debug-seg-fault-in-boostmathlanczos-with-libpcl-surface

commit cfde0b81 (2017-01-16): ad hoc fix to a bug on PCL-1.7 with C++11

fixes #9
closed 2017-01-16 by at-wat