at-wat/mcl_3dl #58: segmentation fault (core dumped) when run mcl_3dl mcl_3dl

weisongwen, 2017-04-28:
First, Thanks for your excellent work!

I catkin_make your package in my workplace successful, but when I run mcl_3dl mcl_3dl, it terminal indicates that  "**segmentation fault (core dumped)**".  my configuration is Ubuntu14.04+ROS indigo.

Also I checked the issue and do as the way in [http://stackoverflow.com/questions/26346690/debug-seg-fault-in-boostmathlanczos-with-libpcl-surface] but it do not work.

when I run **roslaunch mcl_3dl test.launch use_pointcloud_map:=false use_cad_map:=false use_bag_file:=true bag_file:=${HOME}/Downloads/short_test.bag**, the terminal indicates that **[mcl_3dl-4] process has died [pid 8762, exit code -11, cmd /home/wws/5_navigation/devel/lib/mcl_3dl/mcl_3dl ~/cloud:=/cloud ~/mapcloud:=/mapcloud ~/odom:=/odom ~/imu:=/imu/data ~/initialpose:=/initialpose __name:=mcl_3dl __log:=/home/wws/.ros/log/3a977640-2bbd-11e7-9d57-58fb8450c62f/mcl_3dl-4.log].
log file: /home/wws/.ros/log/3a977640-2bbd-11e7-9d57-58fb8450c62f/mcl_3dl-4*.log**

I have Google constantly but I still did not handle it.

at-wat, 2017-04-28:
Could you run mcl_3dl on gdb to find the problem?
http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB will help.

weisongwen, 2017-04-28:
Thanks! I will check it..

at-wat, 2017-05-11:
This can be a combinational bug of c++11, PCL-1.7 and compiler optimization.
In this case, the node fails at boost::math::lanczos::lanczos_initializer.

Please try
```
$ catkin_make -DCMAKE_BUILD_TYPE=Release
```
or use PCL-1.8.

weisongwen, 2017-05-12:
Thanks a lot! excellent answer

weisongwen, 2017-05-12:
To be honestly, .ply file containing PointCloud2 information is too big when localizing in big environments. These days, I find that octomap is a good way to represent the environments and localize in the octomap using Humanoid_localization node, have you tried the octomap? 

Thanks a lot!

at-wat, 2017-05-21:
This node internally uses voxel filter and kd-tree. 
Once the node receives and pre-processes map cloud, memory and CPU usage gets smaller.
In some environments with smaller memory and CPU, it is better to preliminary reduce the number of the points for example by using pcl_outlier_removal and pcl_voxel_grid.
However, I have not confirmed the performance of the node for a large map (e.g. hundreds of the meters).

For ray casting and data registration, octomap is better than kd-tree, however for likelihood model based matching, kd-tree is not bad. Also, the memory efficiency of the octomap is not so better since the points are on the surface of objects and most of the occupied octree nodes are split to minimum grid size.
(In my test during the development of the node for my own dataset, kd-tree implementation was faster than the octomap one.)

I have not tested humanoid_localization yet, however, according to the article about humanoid_localization, that node seems to require pretty much CPU power.

Anyway, I close the issue about segmentation fault.
Please open new issue for performance if you need.

closed 2017-05-21 by at-wat

weisongwen, 2017-05-21:
Hi, @at-wat , Thanks a lot for your reply!

Auctually, the humanoid_localization node requires much in terms of CPU. when I load a big map based on octomap, the rviz will not be fluent.

Also, I encountered some big problems when test the humanoid_loclaization node using my owm .bag files.

Best regards