at-wat/mcl_3dl #59: accuracy issue

DennisBian, 2017-05-10:
Thank you for the excellent work! Really impressive!!

I have successfully used the mcl_3dl to get localization in my own bag, including imu, odom, velodyne hdl-64 cloud data and pcd map(194M). The accuracy of my map is 20cm, however, the average position error is nearly 30m, could u pls give some advices to promote it?thanks u.

attached file is the loaded png map which generated by cartographer using velodyne hdl-64.
[image]

with regards.

at-wat, 2017-05-10:
> The accuracy of my map is 20cm, however, the average position error is nearly 30m, could u pls give some advices to promote it?thanks u.

30 m means not fixed at all?
Currently, the node seems not availabe for small packet of input cloud. (also see #56)
It looks easy to be fixed, however, unfortunately i don't have time until the end of this month.

DennisBian, 2017-05-10:
Hi, Astushi Watanabe,

thanks 4 ur reply, i will check it.

best wishes

at-wat, 2017-06-12:
Setting `accum_cloud` parameter, which is implemented by #60, to be the number of the packet (PointCloud2 message) within one scan of the LIDAR might fix the behavior for Velodyne.

closed 2017-08-16 by DennisBian