at-wat/mcl_3dl #83: Error arise when catkin_make (undefined reference to `boost::chrono::steady_clock::now())

weisongwen, 2017-10-12:
Hi, @at-wat 

Thanks for your work on this package, I encountered an issue bellow when I catkin_make this package. do you have any idea about this question?

Thanks a lot.

CMakeFiles/mcl_3dl.dir/src/mcl_3dl.cpp.o: In function `MCL3dlNode::cbCloud(boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&)':
mcl_3dl.cpp:(.text._ZN10MCL3dlNode7cbCloudERKN5boost10shared_ptrIKN11sensor_msgs12PointCloud2_ISaIvEEEEE[_ZN10MCL3dlNode7cbCloudERKN5boost10shared_ptrIKN11sensor_msgs12PointCloud2_ISaIvEEEEE]+0xd17): undefined reference to `boost::chrono::steady_clock::now()'
mcl_3dl.cpp:(.text._ZN10MCL3dlNode7cbCloudERKN5boost10shared_ptrIKN11sensor_msgs12PointCloud2_ISaIvEEEEE[_ZN10MCL3dlNode7cbCloudERKN5boost10shared_ptrIKN11sensor_msgs12PointCloud2_ISaIvEEEEE]+0x3eeb): undefined reference to `boost::chrono::steady_clock::now()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/wenws/19_mcl3d/devel/lib/mcl_3dl/mcl_3dl] Error 1
make[1]: *** [mcl_3dl/CMakeFiles/mcl_3dl.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

at-wat, 2017-10-12:
Could you post your ROS distribution, OS version (Ubuntu version) and boost version?
Fot Ubuntu, try this:
```
$ echo $ROS_DISTRO
$ cat /etc/lsb-release
$ cat /usr/include/boost/version.hpp | grep "BOOST_LIB_VERSION"
```

This repository is currently automatically tested on OSRF official docker image of ROS kinetic on Ubuntu 16.04.

weisongwen, 2017-10-13:
Hi, @at-wat  thanks for your kind reply! 
This is all the version result bellow!

command: echo $ROS_DISTRO  
**Jade**

command: DISTRIB_ID=Ubuntu
**DISTRIB_RELEASE=14.04
DISTRIB_CODENAME=trusty
DISTRIB_DESCRIPTION="Ubuntu 14.04.4 LTS"**

command: cat /usr/include/boost/version.hpp | grep "BOOST_LIB_VERSION"
**//  BOOST_LIB_VERSION must be defined to be the same as BOOST_VERSION
#define BOOST_LIB_VERSION "1_54"**

command: cat /usr/include/boost/version.hpp | grep "BOOST_VERSION"
**#ifndef BOOST_VERSION_HPP
#define BOOST_VERSION_HPP
//  BOOST_VERSION % 100 is the patch level
//  BOOST_VERSION / 100 % 1000 is the minor version
//  BOOST_VERSION / 100000 is the major version
#define BOOST_VERSION 105400
//  BOOST_LIB_VERSION must be defined to be the same as BOOST_VERSION**

at-wat, 2017-10-16:
Could you try `fix-build-on-indigo` branch?

weisongwen, 2017-10-16:
@at-wat  should I use `fix-build-on-indigo`, as my ros version is **jade**

at-wat, 2017-10-16:
The repository is tested on active LTS ROS distributions (indigo and kinetic) now.
The problem seems occurring on Ubuntu Trusty. Jade and Indigo are both based on Ubuntu Trusty; the fix should also work for Jade.

BTW, Jade has been EOLed on this May. (http://wiki.ros.org/Distributions)

at-wat, 2017-10-20:
I have tested locally on the docker Jade environment.
The update has been merged into master.

Please reopen if the problem still exists.

closed 2017-10-20 by at-wat

weisongwen, 2017-10-21:
@at-wat  **cakin_make** is sucessful now, Thanks for your kind advice and reply!
May you have a nice weekend!