at-wat/mcl_3dl #144: no-odom no-imu version

sunliamm, 2018-05-01:
By any chance, could you provide an example without the usage of odometry and imu? That is only use the 3D Lidar scans for localization.

Thanks!

at-wat, 2018-05-03:
I guess it is not feasible with the current version.
Expanding particle state from 6-dof pose (current impl.) to 12-dof pose+twist might help to realize it.

sunliamm, 2018-05-03:
Thanks! Is there a possibility to use vehicle odometry + Lidar without imu?
We have some experimental data that imu is not available....

at-wat, 2018-05-03:
I think LIDAR with odometry would be ok.

In this node, only acceleration part of the imu data is used for roll and pitch pose estimation.
So, passing acceleration of `(0.0, 0.0, 9.8)` of dummy imu data would work assuming that roll and pitch are not very large for the car.

Note that if you are using Velodyne data, the parameter `accum_cloud` should be `(number of pointcloud messages in one rotation) x N`.
Some more parameters should be modified: e.g. enlarge `clip_*`, `downsample_*`,  `map_downsample_*`, `odom_err_*`. Also, `odom_err_integ_*_sigma` should be very large if the odometry is not accurate.

Currently, there is no document for these a lot of parameters. See: https://github.com/at-wat/mcl_3dl/blob/ccaf2451deead67907d8dd20815c98d791d5cf1b/src/mcl_3dl.cpp#L1472-L1623

sunliamm, 2018-05-03:
> d791d5cf1b/src/mcl_3dl.cpp#L1472-L1623

Thank you indeed for your help! I will give it a try :)

Oyssster, 2021-03-06:
hello, I have tested the version without odom, which means set without_odom:=true. It comes the errors that there is no tf from map to so many links. Is this code can run without odom?Thanks~