at-wat/mcl_3dl #166: initialization of the first epoch

weisongwen, 2018-06-25:
Hi @at-wat ,
Thanks for your great work on this package. Currently, the first position of this package is given manually. but if the initial position is not given accurately, it may fail to initialize. how big the error of the initialization is acceptable in this package?

best,
welson

at-wat, 2018-06-25:
It depends on the number of the particles (`num_particles` parameter) and covariance of the initial pose (`init_var_(x|y|z|roll|pitch|yaw)` parameters or values given by `initialpose` topic).

As another solution, this package has global localization that can be triggered by `/global_localization` service.
Of course, global localization service requires a lot of computation power; it is not reasonable for large maps on the current implementation.

weisongwen, 2018-06-27:
Hi @at-wat ,
thanks for your kind reply, I find that only several beam is used in the beam model of this package.  why you do not use more beams? this is becasue the ray-casting process is quite time-consuming?
best 
welson

weisongwen, 2018-06-27:
hi @at-wat .
sure, do you think the ray-casting in octomap format will be faster?
best 
welson

at-wat, 2018-07-04:
> do you think the ray-casting in octomap format will be faster?

I tried octree in PCL before.
The overhead seems bigger than kd-tree and likelihood computation cost increased a lot.

I haven't tried octomap. It needs complete rewriting of the node, I guess.

at-wat, 2019-12-28:
Closing as the question looks been resolved. Feel free to reopen, or open a new issue, if you have any other questions on this topic.

closed 2019-12-28 by at-wat