at-wat/mcl_3dl #187: Support matching with intensity

at-wat, 2018-08-31:


DaikiMaekawa, 2018-09-20:
looks interesting. you mean, considering intensity a part of matching score in the measurement update, right?

at-wat, 2018-09-20:
Most complex part is treating incidence angle of the beam into the surface and reflection characteristics of it.
Simply taking intensity data into the likelihood calculation makes them improper.
I have some basic ideas to treat them with minimal run-time computation cost.

I'm planning to do it after https://github.com/at-wat/mcl_3dl/pull/195.

greymfm, 2024-02-11:
This could be tricky (treating incidence angle). However, depending on the material, one will get a high intensity (white material), a low (black material) and  something in between (all other). The high and the low is less dependant on the incidence angle.

Idea: Take the difference multiplied by some small weight?  (measurement value - expected value) * weight
Then this could at least filter out particles with wrong material (white vs. black or vice versa).