at-wat/mcl_3dl #222: Error when compile the package

weisongwen, 2019-02-18:
Hi @at-wat , I got a error when compile the latest version of this package,
do you have any ideas on this bug? my computer is : Ubuntu 14.04 + ROS Jade

CMakeFiles/mcl_3dl.dir/src/mcl_3dl.cpp.o: In function `mcl_3dl::MCL3dlNode::cbImu(boost::shared_ptr<sensor_msgs::Imu_<std::allocator<void> > const> const&)':
mcl_3dl.cpp:(.text._ZN7mcl_3dl10MCL3dlNode5cbImuERKN5boost10shared_ptrIKN11sensor_msgs4Imu_ISaIvEEEEE[_ZN7mcl_3dl10MCL3dlNode5cbImuERKN5boost10shared_ptrIKN11sensor_msgs4Imu_ISaIvEEEEE]+0x4a7): undefined reference to `void tf2::doTransform<geometry_msgs::Vector3_<std::allocator<void> > >(geometry_msgs::Vector3_<std::allocator<void> > const&, geometry_msgs::Vector3_<std::allocator<void> >&, geometry_msgs::TransformStamped_<std::allocator<void> > const&)'
collect2: error: ld returned 1 exit status

at-wat, 2019-02-18:
@weisongwen `tf2::doTransform` for `geometry_msgs::Vector3` was [added on Feb 9, 2018](https://github.com/ros/geometry2/commit/21e2d9378742116b9ccb7d2e95ed0963e221b231), and [ROS Jade has been EOL-ed on May 2017](http://wiki.ros.org/Distributions#List_of_Distributions).

No packages are maintained for Jade now; please consider upgrading to live distro.

weisongwen, 2019-02-18:
Hi @at-wat  Thanks for your kind reply. sure.

weisongwen, 2019-02-18:
the package use odometry from /odom topic. however, the topic /tf still contains the transform from /odom to /base_link. this means we need two different odometry?

at-wat, 2019-06-22:
> the package use odometry from /odom topic. however, the topic /tf still contains the transform from /odom to /base_link. this means we need two different odometry?

TF usage in this node is basically same as amcl. (see http://wiki.ros.org/amcl#Transforms)
Localizer estimates `map -> base_link` transform and broadcasts `map -> odom`. This enables other nodes to interpolate short term motion of the robot using odometry.
You can set `~odom_frame` parameter to `base_link` to disable interpolation using odometry and make mcl_3dl node directly broadcast `map -> base_link` transform.
https://github.com/at-wat/mcl_3dl/blob/8d870244b012c8fc77fe85c91536492ec58cee96/config/test_localization_noodom.yaml#L3

`/odom` topic is used to predict particle state transition using twist data in odometry, which is not directly provided in TF.

at-wat, 2019-06-22:
`tf` and `odom` usage is off-topic of this issue.
Feel free to open a new issue if it is still unclear.

closed 2019-06-22 by at-wat