at-wat/mcl_3dl #231: How to use global localization service

ArthurYu96, 2019-06-11:
Thank you for your great works.
I run the demo and it works well, but I don't know how can I use the global localization service, should I write a client for the service?
Btw, if I want to use a real robot without IMU, what should I change?
Hope for your help, thanks

at-wat, 2019-06-14:
Global localization can be triggered by a ROS service.
You can use command line tools `rosservice call /global_localization`.

> Btw, if I want to use a real robot without IMU, what should I change?

This topic is duplicate of #223.
I have not yet decided how to support no-IMU system.

at-wat, 2019-10-24:
An example command to run global localization in the demonstration has been added to README. (https://github.com/at-wat/mcl_3dl/pull/238)

Feel free to reopen if you have question on this topic.

closed 2019-10-24 by at-wat