at-wat/mcl_3dl #249: Why mcl_3dl will stop if using tf static_transform_publisher with x_offset?

HappySamuel, 2019-08-13:
Hi

I set up a static_transform_publisher using tf2_ros for the "base_link -> 3d_lidar" transform. I also got the "odom -> base_link" provided by my gyro_odometry. However, with the following setup:
[image]

I tested with having x_offset with any value, will result the mcl_3dl to run for a while, then stop abruptly without given any error. However with z_offset to be any value, the mcl_3dl wont have this issue.

Do you know why? I tested it with the real robot, and found this issue.

Best,
Samuel

at-wat, 2019-10-24:
Do you mean
- static transform of `0 0 0.45` doesn't works
- static transform of `0.54 0 0.45` works

?

I couldn't reproduce the problem by applying some patterns of laser frame static transforms.
Could you provide `rosrun tf view_frames` result and/or your full launch configurations?

at-wat, 2019-12-28:
Closing as the problem can't be reproduced.
Feel free to reopen, or open a new issue, if you have any another question on this topic.
Attaching a full configuration which reproduces the problem would be great.

closed 2019-12-28 by at-wat