at-wat/mcl_3dl #253: mcl_3dl using which model as default? likelihood model or beam model?

HappySamuel, 2019-08-30:
Hi

I am curious, if i specify neither likelihood nor beam parameters, which is the default model that mcl_3dl is going to use?

Best,
Samuel

at-wat, 2019-08-30:
Both are used. You can change number of the points used in each model separately.

HappySamuel, 2019-09-01:
Hi @at-wat 

Having both to work at the same time? Won't they contradict each other? Can you share the idea of using both of them at the same time?

Besides, if i only want to use one of them, how shall i disable another one.

Best,
Samuel

at-wat, 2019-09-02:
> Having both to work at the same time? Won't they contradict each other?

It's very natural idea to fuse multiple measurement results (likelihood fields) by Bayesian inference.

> Can you share the idea of using both of them at the same time?

See the document linked from README.

> Besides, if i only want to use one of them, how shall i disable another one.

As I explained in the first response, you can set the number of the points used in each model. Beam model can be disabled by setting 0. Disabling likelihood field model is currently not supported.

HappySamuel, 2019-09-04:
Hi @at-wat 

Thanks for the information. I have read the README again. There's another question, the pre-filtered 3D pointcloud, you filtered those whom are useless points (x=0, y=0, z=0)?

Best,
Samuel

at-wat, 2019-10-24:
> There's another question, the pre-filtered 3D pointcloud you mentioned in the README, you filtered those whom are useless points (x=0, y=0, z=0)? or what did you filtered?

No, the filter applied to the sample data is mainly for removing shadow points (by a filter similar to http://wiki.ros.org/laser_filters#ScanShadowsFilter) and points measuring the robot body, which is highly depends on the robot configuration and independent from mcl_3dl algorithm.

at-wat, 2019-12-28:
Closing as the question looks resolved. Feel free to reopen, or open a new issue, if you have any another question on this topic.

closed 2019-12-28 by at-wat

HappySamuel, 2020-01-02:
Ya, the problem already resolved. Sorry for forgetting to close it down. Thanks for the guidance.