at-wat/mcl_3dl #263: Little matched points when running localization

iceicehyhy, 2019-10-23:
Hi there,

I'm trying the localization package using imu, odom and Lidar. But, it seems that the matched points is very little when I move the robot around(less than 10 points). 
My setting is as follows:
1. The map used is a PCD map, and I use pcl_ros pkg to convert it to pointcloud format.
2. Lidar is a 16-channel 360 degree LIDAR.
3. 

I noticed that there're tens of matched points in your demo video, and the localization is accurate. But, in my case, the robot sometimes gets lost and cannot recover due to insufficient matched points.
Any thought on that? Thanks a lot in advance.

at-wat, 2019-10-24:
I guess you are using 3D LIDAR like Velodyne.
Such kinds of LIDARs provide pointcloud in which its horizontal scan is split into several pointcloud messages.
In this case, the `~/accum_cloud` parameter should be increased to the number of the pointcloud messages reconstructing a full horizontal scan.

HappySamuel, 2020-03-07:
Hi @at-wat 

What actually does this accum_cloud help with the localization result for a moving robot?

Best,
Samuel