at-wat/mcl_3dl #266: without imu can not successly run mcl_3dl

txshang, 2019-12-19:
Hello,when I launch the mcl_3dl without imu.It pose will never change, it just stop on a position.
I use test_localization_noodom.yaml
and I publish a static tf from basefootprint to velodyne 
I can use the without_imu version.
Do I do something wrong so that cause this mistake.

at-wat, 2019-12-27:
For systems without IMU nor odometry, using this node is not resonable as it is designed to be lightweight especially on mobile robot system. (6-DoF localization only by using LIDAR requires much amount of computation power.)

At least one of odometry and IMU is mandatory.
https://github.com/at-wat/mcl_3dl/blob/3b6d40c85d1bb5faacd3ef4588caa754b75caec5/src/mcl_3dl.cpp#L1090-L1096

at-wat, 2019-12-28:
Closing as the issue looks a duplicate of #144.
Feel free to reopen if it is not duplicate.

closed 2019-12-28 by at-wat