at-wat/mcl_3dl #276: Reference for motion prediction model

tuandle, 2020-01-09:
Hi @at-wat ,
Thank you for your work. I have a question about the implemented motion model in this package.
The implementation of the diff drive model here seems to be a bit different from what described in Thrun's book and the original ROS navigation pkg.  
I wonder if you can provide some references to how you derive formulas in functions `setOdoms` and `predict`?
I am also confused about the usage of odom error constants, e.g `odom_err_integ_lin_tc_`, `odom_err_integ_ang_tc_` , `noise_ll_`, `noise_la`, etc., can you explain what they are for?
Thanks for your help!

at-wat, 2020-01-10:
> The implementation of the diff drive model here seems to be a bit different from what described in Thrun's book and the original ROS navigation pkg.

I don't have specific reference for it.
Basically, only thing the Bayesian filter requires is that the ground truth state keeps having enough probability, so you don't need to 100% follow the book.
Basic things `predict` and `setOdoms` doing are updating the pose of each particle by using difference of the odometry pose with noise specified by the parameters.

> I am also confused about the usage of odom error constants, e.g odom_err_integ_lin_tc_, odom_err_integ_ang_tc_ , noise_ll_, noise_la, etc., can you explain what they are for?

`odom_err_integ_*` means: https://github.com/at-wat/mcl_3dl/pull/99#issuecomment-363401362

https://github.com/at-wat/mcl_3dl/blob/15f4330b2174d30cabe75d655c6eba61620e05ad/src/mcl_3dl.cpp#L1219-L1222
These `odom_err_*_*` have basically same meaning as amcl's `odom_alpha3`, `odom_alpha2`, `odom_alpha4`, and `odom_alpha1` respectively. (http://wiki.ros.org/amcl#Parameters#line-395-2)

tuandle, 2020-01-10:
@at-wat I got it! Thanks for your reply!

closed 2020-01-10 by tuandle