at-wat/mcl_3dl #280: Particles always converge to the same position

txshang, 2020-01-17:
When I run the code, the particles will always converge to a same position,the map is a pointcloud map with height, so i want to know if some parameters should be set.

at-wat, 2020-01-17:
Information to reproduce and/or describe your problem should be provided. No one can help you without information.

- OS and ROS distribution version
- mcl_3dl version (from source or apt-get)
  - if from source, how did you build (catkin_make/catkin/catkin_make_isolated and optional arguments if specified)
  - if from source, dependent library versions (PCL, Eigen) if changed from rosdep or apt-get provided version
- Information to reproduce your problem, for example:
  - configurations, if it differs from default or demonstration
  - launch files, if it differs from default or demonstration
  - data
- Information to show what exactly happens, for example:
  - screenshots
  - console log/error messages
  - data

txshang, 2020-01-18:
I use ros kinetic on ubuntu16.04

txshang, 2020-01-18:
I use ros kinetic on ubuntu 16.04

txshang, 2020-01-18:
I use ros kinetic on ubuntu16.04
I use catkin_make to build mcl_3dl from source,I use pcl 1.8.1
I don't change default configurations,i changed the launch like this
[image]
I use rslidar points odom and a imu
When I call /global_localization,the position of particles always converge to the same place like this
[image]
Can I give my bag and pcd map to you,so you can give me some help

at-wat, 2020-01-18:
Did you confirmed that the pointcloud, odometry, IMU data are correctly fed to mcl_3dl node?

```xml
  <remap from="rslidar_points" to="rslidar_points" />
```
This has no effect.
If the pointcloud is published to `/rslidar_points`, it is not fed to `mcl_3dl`.

Please check that the topics of pointcloud, odometry, IMU are connected to `mcl_3dl` node. (for example by using `rqt_graph`)

txshang, 2020-01-18:
I have changed some topics name，so the /rslidar_points is no problem，imu and odom are all correct，the problem is just happened on this map and bag，if a map with height is not support

at-wat, 2020-01-18:
https://github.com/at-wat/mcl_3dl/issues/280#issue-551384923
> the map is a pointcloud map with height

https://github.com/at-wat/mcl_3dl/issues/280#issuecomment-575893026
> if a map with height is not support

I couldn't understand what do you mean these part.
As this package is 3-D/6-DOF, the map pointcloud always must have height (z position) information.
Could you explain bit more detail?

txshang, 2020-01-18:
I mean if some configurations should be set，so to achieve a 6DOF localization.Because when I run localization on a robot which always move on a fixed height，it works.but if robot move on terrain , it not works.

txshang, 2020-01-18:
Can I send the bag and map to you, could you do me a favor to test it when you are free

at-wat, 2020-01-18:
Sure, seeing the actual data would be easier.
Could you upload the bag and map to somewhere? or, do you want not to make them public?

txshang, 2020-01-18:
I can upload the data to GitHub and give you a link