at-wat/mcl_3dl #323: Help needed - How to enable taking further feature points into computation

HappySamuel, 2020-06-19:
Hi

When my robot is far away from the building features, the scan-to-map matching become less depend on them, which give a bad localization result. Is there a parameter to adjust for that? Or how shall i amend the code to achieve that? 

Best,
Samuel

at-wat, 2020-06-21:
https://github.com/at-wat/mcl_3dl/blob/b192ce82cee92ff00d21c39e432bc1d560ea2c3a/src/parameters.cpp#L58-L74

`likelihood/clip_far` and `beam/clip_far` should be increased to use the points far from the robot.
`likelihood/match_dist_min` may also be needed to be increased for large environment. Points which are not within `likelihood/match_dist_min` from the map will be ignored in the likelihood calculation.

HappySamuel, 2020-09-01:
Hi @at-wat 

For example, 
for `likelihood/clip_far` as 50m, how much shall i set for `likelihood/match_dist_min` ?
for `likelihood/clip_far` as 100m, how much shall i set for `likelihood/match_dist_min` ?