at-wat/mcl_3dl #355: How to run mcl_3dl with multiple lidars (pointcloud2) ?

HappySamuel, 2020-10-26:
Hi @at-wat 

I just noticed the description of the mcl_3dl is performing a probabilistic 3-D/6-DOF localization system for mobile robots with 3-D LIDAR(s).

How shall i enable it to run with multiple 3D lidars?

Best,
Samuel

at-wat, 2021-01-01:
mcl_3dl subscribes multiple 3D pointclouds (can have multiple frame_id) from `/cloud`.
Transform of multiple frame_ids are handled inside mcl_3dl, so you can just supply all pointclouds into one topic.

HappySamuel, 2021-01-07:
Hi @at-wat 

How will multiple 3D pointclouds data affect the localization result?

Best,
Samuel

flynneva, 2021-02-01:
hey @at-wat, running into the same question over here. how can we put multiple lidars into the same topic as you suggest? is it only a launch file change?

at-wat, 2021-02-02:
For example in the launch file
```xml
<node name="lidar_3d_driver1" ...>
  <remap from="original_topic" to="cloud" />
</node>
<node name="lidar_3d_driver2" ...>
  <remap from="original_topic" to="cloud" />
</node>
```
or
```xml
<node name="lidar_3d_driver1" ... /><!-- publishing to cloud1 -->
<node name="lidar_3d_driver2" ... /><!-- publishing to cloud2 -->

<node pkg="topic_tools" type="relay" name="relay1" args="cloud1 cloud" />
<node pkg="topic_tools" type="relay" name="relay1" args="cloud2 cloud" />
```

at-wat, 2021-02-02:
@HappySamuel mcl_3dl accumulates multiple scans from multiple lidars and process them taking their poses at the timestamp of each message into account.

HappySamuel, 2021-02-02:
Hi @at-wat 

Can it take in the pointcloud data from 3D pointcloud camera as well?

Best
Samuel

at-wat, 2021-02-02:
> Can it take in the pointcloud data from 3D pointcloud camera as well?

I have never tried.