at-wat/mcl_3dl #359: How to compile mcl_3dl in Debug mode ?

IaroslavS, 2020-11-05:
Hi, everyone!
I'm trying to output messages from ROS_INFO, that were included in mcl_3dl code. Also, I want to use ROS_DEBUG_ONCE.
I use VSCode.
To compile in Debug mode I tried the following command:
```
catkin_make -DCMAKE_BUILD_TYPE=Debug
```
But this didn't help: terminal doesn't output messages from ROS_DEBUG and VSCode cannot stop at breakpoints.
I also tried to change mcl_3dl/CMakeLists.txt at line 61 from original:
```
  set(CMAKE_BUILD_TYPE Release)
```
to new modified:
```
  set(CMAKE_BUILD_TYPE Debug)
```
But this also didn't help.
How, you guys, debug mcl_3dl ?

at-wat, 2021-01-01:
See https://answers.ros.org/question/11076/edit-logger-levels-programmatically-or-from-the-command-line/ and http://wiki.ros.org/rosconsole#Changing_Logger_Levels

at-wat, 2021-02-26:
This seems be a general ROS question. See the links above.

closed 2021-02-26 by at-wat