at-wat/mcl_3dl #391: ros2 support

Sollimann, 2022-02-08:
Hi,

First of all, thank you for your great work and contribution.
I was just wondering if there is planned for adding ros2 support?

All Best

at-wat, 2022-02-08:
I don't have plan at now.
Contributions are very welcome! (but the design should be discussed before implementation)

greymfm, 2024-08-02:
There exist (semi-)automated ROS1->ROS2 conversion tools:  https://github.com/awslabs/ros2-migration-tools
 Maybe useful...

inkccc, 2026-01-01:
你好 请问现在有ros2版本的支持了吗

at-wat, 2026-04-01:
In my opinion, the core logic should be separated to a pure C++ library first and a ROS 2 wrapper using the core logic library should be added, instead of directly converting the current ROS 1 code to ROS 2.

See https://github.com/at-wat/mcl_3dl/issues/433 for the details.