at-wat/mcl_3dl #433: Split ROS independent library to support various middleware

at-wat, 2025-07-07:
Split core algorithm as a pure CMake project.
Each middleware can be supported by adding wrappers.

greymfm, 2025-09-03:
Are you working on this already? (If so, can you contact me) - I think I will start implementing this idea (I think it's great)

at-wat, 2025-09-03:
Not working yet. Library API design should be discussed here before starting implementation

greymfm, 2025-09-04:
With some AI assitance, I think I can get a prototype quickly implemented and tested- how about these ideas:

mcl_3dl is already nicely modular (measurement models, motion models, samplers, raycasts), but a lot of it is wired through ROS types (e.g., ros::Duration in Parameters) and node-level concerns. Below is a middleware-agnostic core API you can extract into mcl3dl_core and then keep thin wrappers for ROS 1/ROS 2 (or anything else).

Core package layout (no ROS deps)
'''
mcl3dl_core/
  include/mcl3dl/                # pure C++ headers only depend on Eigen, PCL (optional), std
    types.hpp                    # Pose, Timestamp, Point, CloudView, Covariance
    rng.hpp                      # RNG abstraction
    map.hpp                      # Map/DistanceField/Raycast interfaces
    particle_set.hpp             # Particle container, accessors
    motion_model.hpp             # MotionModel interface + OdometryDelta struct
    sensor_model.hpp             # SensorModel interface (LiDAR/IMU/etc.)
    resampler.hpp                # Resampler (systematic/low-variance), KLD optional
    accumulator.hpp              # Cloud accumulation logic (windowing, downsample)
    options.hpp                  # Options struct + YAML loader (no ros::Duration)
    diagnostics.hpp              # Stats/health
    localizer.hpp                # Localizer (the orchestrator)
  src/
    # concrete impls moved from mcl_3dl:
    lidar_likelihood_model.cpp   # from LidarMeasurementModelLikelihood
    lidar_beam_model.cpp         # from LidarMeasurementModelBeam
    imu_gravity_model.cpp        # from ImuMeasurementModelGravity
    diff_drive_motion.cpp        # from differential drive model
    chunked_kdtree_map.cpp       # from ChunkedKdtree / distance field
    raycaster.cpp                # from raycasts/*
    resampler_systematic.cpp
    cloud_accumulation.cpp
    localizer.cpp
  cmake/…
'''

Minimal type vocabulary
'''
// types.hpp
namespace mcl3dl {
struct Timestamp { int64_t nanoseconds; };              // wrapper, no ROS time
struct Pose6d { Eigen::Vector3d t; Eigen::Quaterniond q; };
struct PoseWithCovariance { Pose6d pose; Eigen::Matrix<double,6,6> cov; };

struct PointXYZI { float x,y,z,i; };
using CloudView = gsl::span<const PointXYZI>;           // or pcl::PointCloud<...>::ConstPtr if you keep PCL
struct Extrinsics { Eigen::Vector3d t; Eigen::Quaterniond q; };
}
'''

Core interfaces (stable across all wrappers)
'''
// map.hpp
struct Ray { Eigen::Vector3d o, dir; double max_range; };
class DistanceField {
public:
  virtual ~DistanceField() = default;
  virtual double distance(const Eigen::Vector3d& p) const = 0;   // nearest-surface distance
  virtual bool raycast(const Ray& r, double* hit_dist) const = 0;
};
class Map {
public:
  virtual ~Map() = default;
  virtual const DistanceField& distanceField() const = 0;
  virtual Eigen::AlignedBox3d bounds() const = 0;
};

// motion_model.hpp
struct OdometryDelta { Eigen::Vector3d dtrans; Eigen::Vector3d drot_rpy; Timestamp stamp; };
class MotionModel {
public:
  virtual ~MotionModel() = default;
  virtual Pose6d sample(const Pose6d& prev, const OdometryDelta& u, std::mt19937& rng) const = 0;
[... 52 lines cut ...]
  void resampleIfNeeded();
  Estimate estimate() const;
  const std::vector<Pose6d>& particles() const;                 // for debug/visualization
  void triggerGlobalLocalization(const Eigen::AlignedBox3d& region, int grid=1);
};
'''

How this maps to current mcl_3dl code
-Sensor models → extract LidarMeasurementModelLikelihood / Beam and ImuMeasurementModelGravity into SensorModel implementations. (The class list already exists under mcl_3dl::….) 
[ROS Documentation](https://docs.ros.org/en/api/mcl_3dl/html/namespacemcl__3dl.html?utm_source=chatgpt.com)
-Motion model → move the differential drive model under MotionModel. (README says only differential drive is supported today.) 
[GitHub](https://github.com/at-wat/mcl_3dl)
-Map/Distance field → wrap ChunkedKdtree/distance field + raycasts behind Map/DistanceField. (Those classes exist today.) 
[ROS Documentation](https://docs.ros.org/en/api/mcl_3dl/html/namespacemcl__3dl.html?utm_source=chatgpt.com)
-Accumulators, samplers, noise → lift CloudAccumulationLogic*, point_cloud_random_samplers/*, and noise_generators/* into core helpers. 
-Parameters → replace mcl_3dl::Parameters (which uses ros::Duration) with Options using plain types/YAML.

Thin wrappers per runtime (keep parity)
-ROS 1/ROS 2 wrappers provide:
-I/O: convert sensor_msgs/PointCloud2, nav_msgs/Odometry, sensor_msgs/Imu, TF → LidarFrame, OdometryDelta, Extrinsics.
-QoS / sync: message_filters (ROS 1) vs. message_filters/rclcpp::Synchronizer (ROS 2) hidden in the wrapper.
-Params: expose the same parameter keys/topics in both wrappers; load into Options.
-Services: map /global_localization → Localizer::triggerGlobalLocalization.
-TF publishing/rviz markers: wrapper-only; core never publishes TF.
-CLI/desktop runner**: read YAML + PCD/PLY map and stream frames from a folder/rosbag for tests and CI.

at-wat, 2025-09-10:
@f-fl0 FYI